Miceuz/rs485-moist-sensor #2: python lib has wrong register numbers

impellerturn, 2018-08-13:
Hi,
thanks for your great work.
i love your sensors.
Anyhow i think that there is a bug with the python implementation installed via pip and also in this repo.
1. The register numbers found in /utils/lib/chirp_modbus.py  are not the same as in https://www.tindie.com/products/miceuz/modbus-rs485-soil-moisture-sensor-2/
2. Also reading the Firmware versions seems not to work, i get IO errors.

I discovered because i was not able to set the measurement interval (and i have my USB dongle on another tty than you and scanning has it hardcoded -> you may want to change that).

here is my version of the file (tty still hardcoded..):
[chirp_modbus.zip](https://github.com/Miceuz/rs485-moist-sensor/files/2283467/chirp_modbus.zip)

You may want to take a look at it, best regards :)

impellerturn, 2019-02-12:
Hello @Miceuz maybe you want to take a look at this issue, too? Thanks and best regards

Miceuz, 2019-03-03:
Thanks for the catch! FW version is from a bit later firmware, so the one you have probably does not have it yet.

commit 9da8622f (2019-03-03): Update chirp_modbus.py

Fix resigter numbers in the python library. Close #2
closed 2019-03-03 by Miceuz