Miceuz/rs485-moist-sensor #4: A few questions about the ATTiny441 programming

cheeeeee, 2019-02-08:
If you get a chance I had a few questions about some of the modbus programming in modbus.c and main.c
# Possible problems:
### In modbus.c --

#### Function Problems:
`bool validateFrame()`
Line 169 -- `uint8_t requestedAdr = getRequestedAddress();`  `requestedAdr` is never used in this function.
`ISR(UART_RECEIVE_INTERRUPT)`
Line 210 -- `uint8_t parityError = (UCSR0C & _BV(UPM01) && (UCSR0A & _BV(UPE0)));`never used in this function.

### In main.c --
#### General Problems:

`F_CPU`
This is defined by default in delay.h as 1MHz and defines the clock cycle. This is probably undefined behavior.

Line 193 -- `ISR(WDT_vect)`
WDT_vect isn't supported on the ATTiny44 according to [https://www.nongnu.org/avr-libc/user-manual/group__avr__interrupts.html](https://www.nongnu.org/avr-libc/user-manual/group__avr__interrupts.html). This is supported if you use Atmel's special header file iotn441.h
This might explain the comment saying you were having problems with it.

#### Function Problems:
`modbusGet(void)`
Line 123 -- `if (modbusIsFrameAvailable())` when true this enters a switch case statement with a default behavior, however there is no error returned if `modbusIsFrameAvailable()` returns false.
Between line 160 and 161 a modbus error should probably be returned to handle a false boolean.

`sleep()`
Line 234 -- might not be a problem but I think you have an unneeded compare.
on line 228 you have `secondsToSleep = holdingRegisters.asStruct.sleepTimes;`
since you've assigned these values to be equal when you do the compare on line 235 `while((sleepTimes > 0) || (secondsToSleep > 0))` I'm pretty sure you're just checking two numbers that are always the same in this particular function.  You can probably use if and just check one.

Line 297 -- incorrect primitive for main
```
void main (void) {
}
```
should be 
```
int main (void) {
}
```

Lines 318 and 324

Incorrect primitives for temp.  Needs to be signed to allowed for negative values. Initial declaration is correct but is incorrect in function call.

`(uint16_t*) &(inputRegisters.asStruct.temperature));`
should be
`(int16_t*) &(inputRegisters.asStruct.temperature));`

# Questions!
1. What do you think the F_CPU cycle should be set to? I'm assuming 16MHz to match the crystal but it might be 8 MHz to match the ATTiny441 default clock.
2.  Currently if I write to a value to the sleep register it won't wake up until that counter expires or the sensor is power cycled.  Is this intentional or a bug? It seems like requesting a read operation should wake the sensor.  I think with the  iotn441.h header fix this could be easily done with a watchdog interrupt.

You can find everything I've done to these files at [https://github.com/cheeeeee/RS485_AtmelDebug](https://github.com/cheeeeee/RS485_AtmelDebug) Including some additional API documentation in main.c if you haven't looked at this in a while and need a refresher.

cheeeeee, 2019-02-15:
I answered question 1 for myself.  Turns out iT was defined in the original makefile as 16MHz and my compiler was giving me an undefined error because Atmel studio was managing my makefile for me.

Miceuz, 2019-03-03:
Hi, thanks for your feedback.

F_CPU is defined in Makefile

This is attiny441, not attiny44, avr-libc was lagging behind a bit.

modbusGet is polled from the mail loop, it does not result in error if there is no modbus frame to process, it simply did not come yet.

You are right about double nested sleep loop - something is fishy there. But you are mixind two things sleepTimeS means "sleep time in seconds" while sleepTimes means how many times sleep was performed.

Also you are right regarding the pointer type for the temperature -- will fix that.

Tha fact that sensor is not accessible during sleep is by design as I want to save the power by putting the rs485 transceiver to shutdown also. I do that by disabling both reader and driver.

Thanks again for you contribution!

cheeeeee, 2019-03-05:
Yep. I was a little confused since the astruct and variable had the same name. I clarified that in my pull request if you don't mind it.  Thanks for getting back to me.  Hopefully I can flash the firmware on my sensor with my updated code to make sure it's gonna work when I get a chance to.

closed 2019-03-05 by cheeeeee