Miceuz/rs485-moist-sensor #6: RS485 Bus stops after a few days

impellerturn, 2019-02-12:
Hello,
i´ve encountered, that my sensors are not working after a few days of operation.
### Setup
- 3 Sensors
- overall bus length 25m
- Termination: VCC <-> 390Ohms <-> RS485-line-A <-> 220Ohms <-> RS485-line-B <-> 390Ohms <-> Ground
- Termination is at my USB to RS485 Transceiver

### Measurement
- set measuerement interval to 500ms
- do 100 measuerements, read the data once per second then calculate mean on the data
- then do the next 100 measuerements

### Result
- This runs fine for about 2 days but occasionally stops.
- Tested with the USBtoRS485 converter sold by @Miceuz and also with a different one. Same results.
- Only pwer cycling the whole bus helps.

Has anyone else had similar issues? Maybe @[cheeeeee](/cheeeeee)?

Any help appreaciated, thanks and best regards

cheeeeee, 2019-02-12:
My initial thought might be the watchdog timer on it. It looks like the WDTCSR is still in the works honestly. I don't even have my sensors in production yet since I'm still going through the firmware.  I'll see about looking into improving the watchdog on it and submitting a pull request since Albertas seems pretty busy.  If you're not familiar, a watchdog timer basically monitors for a loop condition and if a timeout is reached it resets the device automatically.  To keep that from happening you have to keep on feeding the watchdog timeout conditions.  Is there a delay between when you do batches of measurements? Can you try a lower number of reads and report back? Like say 50 reads and average and see if it dies in 2 days or something like 4 days?

impellerturn, 2019-02-14:
Hi,
i´ve tried bringing my system up yesterday but something with MQTT did not work out. Will look for it on the weekend.
Here is my code to use the sensors. It requires the mosquitto python module along with some standard python libs. Dont have a public repo yet.

[soilmoisturereader.zip](https://github.com/Miceuz/rs485-moist-sensor/files/2863887/soilmoisturereader.zip)

Best regards

cheeeeee, 2019-02-14:
Okay just fair warning I haven't used python before but everything that I see on your controllers looks good to me.  Most errors are handled and you should be getting exceptions reported if there is an error in the controller querying the modbus devices.  Try using a lower query rate like I suggested before and see if that changes anything.  If it does that might indicate a limit or bug with MQTT. There was a POSSIBLE error with a write protected register in attint441 programming when sleep was called and a reset might not be initiated and an infinite loop would occur. I included that fix with with other request but I have all of 4 weeks of experience with cpp and about 3 weeks of experience writing c so I'm mostly going off of datasheets and research here.  Albertas will really need to check it all and program one and see what happens.  If what I have doesn't fix it we might have to approach it from a different angle. Just wait and see at this point. Hopefully this works.

Miceuz, 2019-03-03:
@cheeeeee reset() in not called from sleep(). 

Sensor is configured in Safety Level 1. 

> In this mode, the Watchdog Timer is initially disabled, but can be enabled by writing the WDE bit to one without any restriction. A timed sequence is needed when disabling an enabled Watchdog Timer. 

@impellerturn are you putting sensors to sleep?

impellerturn, 2019-03-03:
Hi,
no i do not bring the sensors to sleep.

Miceuz, 2021-04-18:
Sorry, I have totally forgotten this. 

My suspicion lies with the powering. How many sensors are powered from a single usb dongle? I wouldn't use USB dongle as a permanent power solution. It can supply only one - two sensors.

impellerturn, 2021-04-19:
There were 3 sensors powered by the usb dongle.
In the first post i've listed a detailed description of the system setup.

As of today i do not have the setup any more.